adafruit/Adafruit-WebIDE #279: Screens do not match - Integrated Applications.

DWalton4, 2015-01-12:
I install WebIDE, navigate to 192.168.x.x/Setup, and see 
"Welcome to the Raspberry Pi WebIDE powered by the Adafruit Learning System"

Create BitBucked account, login, navigate to "Manage Account".
I see "Account Settings" on the left panel.
There is no option visible for "Integrated Applications" as appears in the documentation. 

Do I need to enable an option somewhere to make it visible, or is the documentation outdated and no longer matching the BitBucket screens, or something else?

Thanks

peterrus, 2015-03-22:
Account -> Access Management -> OAuth

SjoerdOptLand, 2016-05-08:
After some fiddling around, I found the page you refer to, @peterrus.

However, the settings are a bit richer than suggested on the install page. For example, I also need to check boxes to grant rights to WebIDE. Which ones?

This seems like a minor update to match BitBucket that could easily be done by someone who has the WebIDE developing toolchain up and running. It may be a showstopper for someone who just starts out with developing Python on a RaspberryPi, however. Priority?

closed 2018-03-06 by jwcooper