adafruit/Adafruit-WebIDE #290: Sample python scripts preloaded.

aigarslv, 2015-06-23:
Heya Adafruit - excellent work on IDE! :+1:  I think it would be great to add sample scripts that use GPIO like blinking a LED. Now I realize that this IDE is more for professional programmers, but Visualize feature makes this ideal environment for learning :)

beruic, 2016-01-14:
Perhaps even a GPIO diagnostics/debugging panel, where you could graphically read the status of the GPIO, and try things out before coding them?

jwcooper, 2018-03-06:
We're closing all existing issues as we have a fairly large update to the webide. Please create a new issue for any issues found once we release the new version. Thanks!

closed 2018-03-06 by jwcooper

beruic, 2018-03-07:
When about can we expect it to be ready?
And what version will it have?

ladyada, 2018-03-07:
There's no ETA - subscribe to this repo or the adafruit blog for any release updates :)