adafruit/Adafruit-WebIDE #292: My-pi-projects not visible in WebIDE

MichelReij, 2015-09-20:
After installation of the WebIDE with a fresh bitbucket account, I don't see the "my-pi-projects" folder and I don't have the option to create new files or folders.

[image]
I suppose I should have set some additional rights somewhere, but I fail to see where. The "my-pi-projects" is present in BitBucket.

mapocosm, 2015-10-29:
I get the same result, running Chrome and Firefox on OSX. On Windows, Chrome couldn't resolve the URL name, and Firefox converted the name to www.raspberrypi.local and failed. Both computers are on the same wired switch as the Raspberry Pi.

codycoder, 2015-11-26:
I have the same issue.

sethianuj, 2016-01-23:
Solution is:
1. go to your bitbucket account, create a repository there.
2. Then, in your bitbucket,org "repository setup" section, click on "I am starting from scratch" and look at "setup your local directory".
3. Come back to WebIDE, and click on "Terminal" (top middle of WebIDE). In the command, create a new directory (this new directory is the equivalent of "/path/to/your/project"),  execute those git commands.
4. Press the refresh button in top-left of webIDE where the repositories are listed.

codycoder, 2016-01-23:
I'm setting up another PI soon, I'll try this way next time.  I found a different solution here which also worked.

https://github.com/adafruit/Adafruit-WebIDE/issues/295#issuecomment-173984465

closed 2018-03-06 by jwcooper