adafruit/Adafruit-WebIDE #295: Cannot clone my-pi-projects into WebIDE from BitBucket.

codycoder, 2015-11-26:
I am using a fresh rPi (raspberrian-jessie) and the latest WebIDE (v0.3.8).  I followed the instructions here:
> https://learn.adafruit.com/webide/installation

Installed via terminal, on the rPi box itself from a windowed session using:
> curl https://raw.githubusercontent.com/adafruit/Adafruit-WebIDE/alpha/scripts/install.sh | sudo sh

The expected results happened, I login and can see the "Adafruit-Raspberry-Pi-Python-Code" but not the my-pi-projects repo.  

I do have that repo in my bit bucket account.  The IDE, when it's loading claims it's checking the "my-pi-projects" however it's not actually present nor does the IDE have permission to get this repo at this stage.

When I manually clone the "my-pi-projects" repo, by selecting it in BB, getting the SSH url and copy and pasting it into the Clone Repo window in the WebIDE I get public key errors.

I looked for the SSH keys and found them here (using the "terminal" view in WebIDE)

> webide@raspberrypi /usr/share/adafruit/webide/repositories $ cd ~                                                              
> webide@raspberrypi ~ $ ls ~/.ssh                                                                                               
> config  id_rsa_bitbucket  id_rsa_bitbucket.pub  known_hosts                                                                    

The BB PUB key is not present in my BitBucket account and I think this causes these errors.

So, I sudo copied  id_rsa_bitbucket.pub from the webide user to the pi user's home directory because I don't know the webide password and I didn't want to enable root.  

Then I ssh'ed over to the pi user from my mac and used "pbcopy" to get the contents into my clipboard, added it to bb and reloaded everything.

Now I have the my-pi-projects repo and I can copy files over from the adafruit.  I created a folder, a file, ran a test file and confirmed that the test file was pushed up to BB.

I think it's working now.

olliman1, 2016-01-22:
I have the same problem, unfortunately I'm not able to understand how you fixed it.

codycoder, 2016-01-22:
I think there are some bugs in the WebIDE project, for sure.

So what I was trying to explain (perhaps badly) was that I needed to get the public SSH key from the webide user account which was created by, and is used by the Web IDE application; over to my BitBucket account because it was missing.  This problem at it's heart, is an authentication problem.

My goal was: to get the public key from the 'webide' user into bit bucket because it was not present in my new bitbucket account.

I'm also assuming basic familiarity with linux. If you are new to linux os' then these instructions are unfortunately harder.  Sorry about that.

1. If you login to bit bucket and look under Settings >> SSH Keys and you don't see any (pertaining to your raspi anyway, you might be using bit bucket for other things) - _then this is your problem_.  You need to get the public key from your 'webide' user INTO your bit bucket account.

2. This was my problem too.  So I looked in the 'webide' user's .ssh folder (this is where ssh keys are kept on linux os' and a dot in front of a folder or file name signifies a hidden resource so you have to make sure you include hidden objects in search results.  SO:

````bash
pi@localhost~# ls -la /home/webide/.ssh 
````

Will show you the contents of the ssh folder for the webide user.  There will be at least two files there.  They are named in pairs; one file is the private key file and the other is the public key and is usually named with a "pub" in it's file name:

````bash
id_rsa_bitbucket << PRIVATE LEAVE ALONE
id_rsa_bitbucket.pub << THE ONE YOU NEED
````

````bash
So an easy way which I released later  is that you can just do:
````

````bash
pi@localhost~# nano /home/webide/.ssh/id_rsa_bitbucket.pub 
````

Which will open a terminal based text editor called nano and will show you the contents of the public key.
If your terminal program allowed you to copy and paste, you can select ALL of the text on the single line of text and COPY it to your computers clipboard (called the pasteboard on mac).

3. Login to BitBucket, go to Settings >> SSH Keys and create a new Public Key.  Name it anything you like and PASTE the contents of your clipboard into the window.

There are many ways of getting at this data.  I've shown two. Both work; the second one is easier than the first.

I hope this helps you :) 

Good luck.  

[image]

[image]

VTDuffman, 2016-01-22:
This just worked for me, it was definitely the problem, the lack of key in bitbucket.

As a note, I had a bear of a time trying to copy the whole key in nano because it went off the screen. I used sudo vi /home/webide/.ssh/id_rsa_bitbucket.pub and it wrapped onto the screen and I could just copy an paste into BitBucket using my mouse. when I reloaded the WebIDE, my-pi-projects appeared automatically.

Thanks for the help.

codycoder, 2016-01-22:
Great news!
I knew that would be a problem lol.  It always is lol

olliman1, 2016-01-23:
@codycoder Your instructions where adequate for my basic linux familiarity, it's now working! Thank you!
@VTDuffman Thank you for that tip, I would not know how to get the whole SSH otherwise.

leinad13, 2017-02-14:
I just had the same issue - thanks for the details on how to fix. Any news from dev's on a planned code fix?

closed 2018-03-06 by jwcooper