adafruit/Adafruit-WebIDE #296: new version won't install, and borked previous install.

MonadicNomad, 2015-12-01:
Chose "upgrade" from the web interface.  It hung at "restarting"  Now I can't install using manual instructions on adafruit.com.  https://learn.adafruit.com/webide/installation

Specifically, issuing the "npm install" command, get "Error: failed to fetch from registry: passport-github/0.1.3"

jwcooper, 2015-12-01:
Are you using debian jessie?

Interesting that you can't npm install, as I was doing that quite a few times yesterday from the fresh installation on debian jessie. What version of node and npm are you using? Try upgrading npm?

You could also try uninstalling, and then re-installing it using the script or the deb file.

MonadicNomad, 2015-12-01:
No.  Still on Wheezy.  

Doesn't the first line of the manual install ...

sudo apt-get update && sudo apt-get -y install build-essential nodejs nodejs-legacy npm redis-server git

... load the latest node and npm?  Sorry.  I'm a linux amateur.

MonadicNomad, 2015-12-01:
build-essential is already the newest version.
git is already the newest version.
nodejs is already the newest version.
nodejs-legacy is already the newest version.
npm is already the newest version.
redis-server is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 5 not upgraded.

Oh WEIRD... tried again and this time it got passport-git but barfed with "Error: failed to fetch from registry: express/3.1.0"

Is registry.npmjs.org messed up?

MonadicNomad, 2015-12-01:
I'm not sure what to do at this point.  After trying a few more times, I'm not sure if anything in the 'npm install' is working.  This is the response:

npm http GET https://registry.npmjs.org/express/3.1.0
npm http GET https://registry.npmjs.org/express-session/1.12.1
npm http GET https://registry.npmjs.org/passport/0.1.15
npm http GET https://registry.npmjs.org/passport-bitbucket/0.1.2
npm http GET https://registry.npmjs.org/passport-github/0.1.3
npm http GET https://registry.npmjs.org/jade/0.28.1
npm http GET https://registry.npmjs.org/request/2.14.0
npm http GET https://registry.npmjs.org/socket.io/0.9.13
npm http GET https://registry.npmjs.org/connect/2.7.3
npm http GET https://registry.npmjs.org/connect-redis
npm http GET https://registry.npmjs.org/redis/0.8.2
npm http GET https://registry.npmjs.org/validator/0.4.25
npm http GET https://registry.npmjs.org/winston/0.6.2

npm ERR! Error: failed to fetch from registry: express/3.1.0
npm ERR!     at /usr/share/npm/lib/utils/npm-registry-client/get.js:139:12
npm ERR!     at cb (/usr/share/npm/lib/utils/npm-registry-client/request.js:31:9)
npm ERR!     at Request._callback (/usr/share/npm/lib/utils/npm-registry-client/request.js:136:18)
npm ERR!     at Request.callback (/usr/lib/nodejs/request/main.js:119:22)
npm ERR!     at Request.<anonymous> (/usr/lib/nodejs/request/main.js:212:58)
npm ERR!     at Request.emit (events.js:88:20)
npm ERR!     at ClientRequest.<anonymous> (/usr/lib/nodejs/request/main.js:412:12)
npm ERR!     at ClientRequest.g (events.js:156:14)
npm ERR!     at ClientRequest.emit (events.js:67:17)
npm ERR!     at HTTPParser.parserOnIncomingClient [as onIncoming] (http.js:1256:7)
npm ERR! You may report this log at:
npm ERR!     <http://bugs.debian.org/npm>
npm ERR! or use
npm ERR!     reportbug --attach /home/pi/Adafruit-WebIDE/npm-debug.log npm
npm ERR! 
npm ERR! System Linux 4.1.7+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install"
npm ERR! cwd /home/pi/Adafruit-WebIDE
npm ERR! node -v v0.6.19
npm ERR! npm -v 1.1.4
npm ERR! message failed to fetch from registry: express/3.1.0
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/Adafruit-WebIDE/npm-debug.log
npm not ok

MonadicNomad, 2015-12-01:
Here's the npm-debug.log.  I hope it's helpful.

info it worked if it ends with ok
verbose cli [ '/usr/bin/nodejs', '/usr/bin/npm', 'install' ]
info using npm@1.1.4
info using node@v0.6.19
verbose config file /home/pi/.npmrc
verbose config file /usr/etc/npmrc
verbose config file /usr/share/npm/npmrc
verbose caching /home/pi/Adafruit-WebIDE/package.json
verbose loadDefaults raspeditor@0.1.2
verbose readDependencies: using package.json deps
verbose where, deps [ '/home/pi/Adafruit-WebIDE',
verbose where, deps   [ 'express',
verbose where, deps     'express-session',
verbose where, deps     'passport',
verbose where, deps     'passport-bitbucket',
verbose where, deps     'passport-github',
verbose where, deps     'jade',
verbose where, deps     'request',
verbose where, deps     'socket.io',
verbose where, deps     'gitty',
verbose where, deps     'jsDAV',
verbose where, deps     'connect',
verbose where, deps     'connect-redis',
verbose where, deps     'redis',
verbose where, deps     'tty.js',
verbose where, deps     'validator',
verbose where, deps     'winston',
verbose where, deps     'later' ] ]
verbose from cache /home/pi/Adafruit-WebIDE/package.json
info preinstall raspeditor@0.1.2
verbose from cache /home/pi/Adafruit-WebIDE/package.json
verbose readDependencies: using package.json deps
verbose cache add [ 'express@3.1.0', null ]
silly cache add: name, spec, args [ undefined, 'express@3.1.0', [ 'express@3.1.0', null ] ]
verbose parsed url { pathname: 'express@3.1.0',
verbose parsed url   path: 'express@3.1.0',
verbose parsed url   href: 'express@3.1.0' }
silly cache add: name, spec, args [ 'express', '3.1.0', [ 'express', '3.1.0' ] ]
verbose parsed url { pathname: '3.1.0', path: '3.1.0', href: '3.1.0' }
verbose addNamed [ 'express', '3.1.0' ]
verbose addNamed [ '3.1.0', '3.1.0' ]
verbose cache add [ 'express-session@1.12.1', null ]
silly cache add: name, spec, args [ undefined,
silly cache add: name, spec, args   'express-session@1.12.1',
silly cache add: name, spec, args   [ 'express-session@1.12.1', null ] ]
verbose parsed url { pathname: 'express-session@1.12.1',
verbose parsed url   path: 'express-session@1.12.1',
verbose parsed url   href: 'express-session@1.12.1' }
silly cache add: name, spec, args [ 'express-session', '1.12.1', [ 'express-session', '1.12.1' ] ]
verbose parsed url { pathname: '1.12.1', path: '1.12.1', href: '1.12.1' }
verbose addNamed [ 'express-session', '1.12.1' ]
verbose addNamed [ '1.12.1', '1.12.1' ]
verbose cache add [ 'passport@0.1.15', null ]
silly cache add: name, spec, args [ undefined, 'passport@0.1.15', [ 'passport@0.1.15', null ] ]
verbose parsed url { pathname: 'passport@0.1.15',
verbose parsed url   path: 'passport@0.1.15',
verbose parsed url   href: 'passport@0.1.15' }
silly cache add: name, spec, args [ 'passport', '0.1.15', [ 'passport', '0.1.15' ] ]
verbose parsed url { pathname: '0.1.15', path: '0.1.15', href: '0.1.15' }
verbose addNamed [ 'passport', '0.1.15' ]
verbose addNamed [ '0.1.15', '0.1.15' ]
verbose cache add [ 'passport-bitbucket@0.1.2', null ]
silly cache add: name, spec, args [ undefined,
silly cache add: name, spec, args   'passport-bitbucket@0.1.2',
silly cache add: name, spec, args   [ 'passport-bitbucket@0.1.2', null ] ]
verbose parsed url { pathname: 'passport-bitbucket@0.1.2',
verbose parsed url   path: 'passport-bitbucket@0.1.2',
verbose parsed url   href: 'passport-bitbucket@0.1.2' }
[... 238 lines cut ...]
silly spawning     '/home/pi/Adafruit-WebIDE/tmp/npm-1449012713793/1449012713839-0.6037093142513186' ],
silly spawning   null ]
ERR! Error: failed to fetch from registry: express/3.1.0
ERR!     at /usr/share/npm/lib/utils/npm-registry-client/get.js:139:12
ERR!     at cb (/usr/share/npm/lib/utils/npm-registry-client/request.js:31:9)
ERR!     at Request._callback (/usr/share/npm/lib/utils/npm-registry-client/request.js:136:18)
ERR!     at Request.callback (/usr/lib/nodejs/request/main.js:119:22)
ERR!     at Request.<anonymous> (/usr/lib/nodejs/request/main.js:212:58)
ERR!     at Request.emit (events.js:88:20)
ERR!     at ClientRequest.<anonymous> (/usr/lib/nodejs/request/main.js:412:12)
ERR!     at ClientRequest.g (events.js:156:14)
ERR!     at ClientRequest.emit (events.js:67:17)
ERR!     at HTTPParser.parserOnIncomingClient [as onIncoming] (http.js:1256:7)
ERR! You may report this log at:
ERR!     <http://bugs.debian.org/npm>
ERR! or use
ERR!     reportbug --attach /home/pi/Adafruit-WebIDE/npm-debug.log npm
ERR! 
ERR! System Linux 4.1.7+
ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install"
ERR! cwd /home/pi/Adafruit-WebIDE
ERR! node -v v0.6.19
ERR! npm -v 1.1.4
ERR! message failed to fetch from registry: express/3.1.0
verbose exit [ 1, true ]

jwcooper, 2015-12-02:
I think this is the issue:

    ERR! node -v v0.6.19
    ERR! npm -v 1.1.4

Those are really old versions of node and npm. Your best bet is to either try the install script again, or run the uninstall, then the install script. Those scripts don't rely on the version of node or npm you have installed, and include all the node dependencies that npm is trying to install.

MonadicNomad, 2015-12-02:
I tried running the curl uninstall/install scripts *repeatedly*.  It says it installs.  But the editor doesn't respond to http gets.

I gave up and installed a fresh image of jessie.  The IDE installed.  I accessed via http and gave it my bitbucket credentials.  Now it just sits there with the following ... It been like this for 10 minutes.  Same thing if I reload.  This is very frustrating.

Initializing Editor...

Initializing Editor Commands

Initializing Editor Events

Populating Editor Bar

Checking Editor Health

jwcooper, 2015-12-02:
Can you try starting it from the command line?

    sudo service adafruitwebide stop
    cd /usr/share/adafruit/webide
    node server

Check out the output and report any errors. If no issues, then navigate to the logs directory and check out those log files and paste any errors here.

Which Raspberry Pi are you using?

MonadicNomad, 2015-12-02:
Nope.  Still stuck at checking editor health.  Here's the last entry in error.log

2015-12-02T16:31:49.854Z - error: uncaughtException date=Wed Dec 02 2015 08:31:49 GMT-0800 (PST), pid=1694, uid=1001, gid=1001, cwd=/, execPath=/usr/share/adafruit/webide/bin/node_hf/node, version=v0.10.29, argv=[/usr/share/adafruit/webide/bin/node_hf/node, /usr/share/adafruit/webide/server.js], rss=43720704, heapTotal=33962496, heapUsed=19779716, loadavg=[0.29150390625, 0.1787109375, 0.10791015625], uptime=4311.049932689, trace=[column=15, file=module.js, function=Function.Module._resolveFilename, line=338, method=Module._resolveFilename, native=false, column=25, file=module.js, function=Function.Module._load, line=280, method=Module._load, native=false, column=17, file=module.js, function=Module.require, line=364, method=require, native=false, column=17, file=module.js, function=require, line=380, method=null, native=false, column=25, file=/usr/share/adafruit/webide/server.js, function=SocketNamespace.auth, line=387, method=auth, native=false, column=15, file=/usr/share/adafruit/webide/node_modules/socket.io/lib/namespace.js, function=SocketNamespace.authorize, line=245, method=authorize, native=false, column=14, file=/usr/share/adafruit/webide/node_modules/socket.io/lib/namespace.js, function=SocketNamespace.handlePacket, line=302, method=handlePacket, native=false, column=32, file=/usr/share/adafruit/webide/node_modules/socket.io/lib/manager.js, function=Manager.handleClient, line=697, method=handleClient, native=false, column=8, file=/usr/share/adafruit/webide/node_modules/socket.io/lib/manager.js, function=Manager.handleUpgrade, line=618, method=handleUpgrade, native=false, column=10, file=/usr/share/adafruit/webide/node_modules/socket.io/lib/manager.js, function=, line=123, method=null, native=false], stack=[Error: Cannot find module 'cookie',     at Function.Module._resolveFilename (module.js:338:15),     at Function.Module._load (module.js:280:25),     at Module.require (module.js:364:17),     at require (module.js:380:17),     at SocketNamespace.auth (/usr/share/adafruit/webide/server.js:387:25),     at SocketNamespace.authorize (/usr/share/adafruit/webide/node_modules/socket.io/lib/namespace.js:245:15),     at SocketNamespace.handlePacket (/usr/share/adafruit/webide/node_modules/socket.io/lib/namespace.js:302:14),     at Manager.handleClient (/usr/share/adafruit/webide/node_modules/socket.io/lib/manager.js:697:32),     at Manager.handleUpgrade (/usr/share/adafruit/webide/node_modules/socket.io/lib/manager.js:618:8),     at Server.<anonymous> (/usr/share/adafruit/webide/node_modules/socket.io/lib/manager.js:123:10)]

ApisMellow, 2015-12-02:
seeing same issue, raspberry pi version 1, hangs forever at checking editor health. Testing in multiple client machines with both Chrome and Safari. Persistent across a reboot of the RPi after installation via the "curl |" method.

jwcooper, 2015-12-02:
Found the issue, and working on a new release right now. Quick fix:

    cd /usr/share/adafruit/webide
    sudo npm install cookie
    sudo service adafruitwebide restart

Sorry for the inconvenience! :(

MonadicNomad, 2015-12-02:
Will this unbork my old wheezy install?

jwcooper, 2015-12-02:
Probably, I'd try it out.

MonadicNomad, 2015-12-02:
sorry.  this must be very basic.  But where is npm?  from 

pi@pypi:/usr/share/adafruit/webide $ sudo npm install cookie

I get ...

sudo: npm: command not found

jwcooper, 2015-12-02:
you can 'sudo apt-get install npm'

MonadicNomad, 2015-12-02:
oh of course.  for some reason i thought it came in your distro.  ok.  cookie installed (i think.  npm was very quiet).  recycled webide service.  can now access through http.  It appears to have cloned my repositories too.  so that looks good.

Now it says that version 0.3.12 is available.  Should I install that, keeping in mind that that's how I got into this trouble?

jwcooper, 2015-12-03:
You can skip that version as it just includes the cookie package.

CheapB, 2015-12-04:
Have the exact same problem. Even tried to uninstall (which btw removed my code repository - not cool) and reinstalled. still same problem. Started fresh on a new jessie image and 2 hours into it I am getting "Error: Failure updating from remote repository" and bitbucket is not updating. good times....

ApisMellow, 2015-12-05:
The instructions you gave above resolved my issue: 

 ssh into the raspberry pi
    
    cd /usr/share/adafruit/webide
    sudo apt-get update
    sudo apt-get install npm
    sudo npm install cookie
    sudo service adafruit-webide restart
    
relog into the web interface again via chrome at http://raspberrypi.local (reload page or use the IP address of the pi instead of the raspberrypi.local URL if it refuses to resolve properly)

Thanks so much for this, the IDE is fantastic! I love the automated commits back so that everyone gets built in benefits of source control!

MonadicNomad, 2015-12-07:
@ApisMellow Please tell me if you're on wheezy or jessie.  Thanks.

2stacks, 2016-01-20:
I'm running wheezy, same issue as @MonadicNomad.  Tried to update, update failed now I can't get a a clean install on wheezy working.  Followed the previous recommendations but I get the following when trying to install 'cookie'.

pi@raspi-01:/usr/share/adafruit/webide$ sudo npm install cookie
npm http GET https://registry.npmjs.org/cookie/0.2.3

npm ERR! Error: failed to fetch from registry: cookie/0.2.3
npm ERR!     at /usr/share/npm/lib/utils/npm-registry-client/get.js:139:12
npm ERR!     at cb (/usr/share/npm/lib/utils/npm-registry-client/request.js:31:9)
npm ERR!     at Request._callback (/usr/share/npm/lib/utils/npm-registry-client/request.js:136:18)
npm ERR!     at Request.callback (/usr/lib/nodejs/request/main.js:119:22)
npm ERR!     at Request.<anonymous> (/usr/lib/nodejs/request/main.js:212:58)
npm ERR!     at Request.emit (events.js:88:20)
npm ERR!     at ClientRequest.<anonymous> (/usr/lib/nodejs/request/main.js:412:12)
npm ERR!     at ClientRequest.g (events.js:156:14)
npm ERR!     at ClientRequest.emit (events.js:67:17)
npm ERR!     at HTTPParser.parserOnIncomingClient [as onIncoming] (http.js:1256:7)
npm ERR! You may report this log at:
npm ERR!     <http://bugs.debian.org/npm>
npm ERR! or use
npm ERR!     reportbug --attach /usr/share/adafruit/webide/npm-debug.log npm
npm ERR! 
npm ERR! System Linux 4.1.13+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "cookie"
npm ERR! cwd /usr/share/adafruit/webide
npm ERR! node -v v0.6.19
npm ERR! npm -v 1.1.4
npm ERR! message failed to fetch from registry: cookie/0.2.3
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /usr/share/adafruit/webide/npm-debug.log
npm not ok

pi@raspi-01:/usr/share/adafruit/webide$ cat  /usr/share/adafruit/webide/npm-debug.log
info it worked if it ends with ok
verbose cli [ '/usr/bin/nodejs', '/usr/bin/npm', 'install', 'cookie' ]
info using npm@1.1.4
info using node@v0.6.19
verbose config file /root/.npmrc
verbose config file /usr/etc/npmrc
verbose config file /usr/share/npm/npmrc
verbose caching /usr/share/adafruit/webide/package.json
verbose loadDefaults raspeditor@0.1.2
verbose caching /usr/share/adafruit/webide/node_modules/cookie/package.json
verbose loadDefaults cookie@0.2.3
verbose caching /usr/share/adafruit/webide/node_modules/request/package.json
verbose caching /usr/share/adafruit/webide/node_modules/validator/package.json
verbose loadDefaults validator@0.4.25
verbose caching /usr/share/adafruit/webide/node_modules/passport-github/package.json
verbose caching /usr/share/adafruit/webide/node_modules/connect-redis/package.json
verbose loadDefaults connect-redis@2.5.1
verbose caching /usr/share/adafruit/webide/node_modules/connect/package.json
verbose loadDefaults connect@2.7.3
verbose caching /usr/share/adafruit/webide/node_modules/later/package.json
verbose loadDefaults later@0.0.5
verbose caching /usr/share/adafruit/webide/node_modules/jade/package.json
verbose loadDefaults jade@0.28.1
verbose caching /usr/share/adafruit/webide/node_modules/winston/package.json
verbose caching /usr/share/adafruit/webide/node_modules/gitty/package.json
verbose loadDefaults gitty@0.0.1
verbose caching /usr/share/adafruit/webide/node_modules/tty.js/package.json
verbose loadDefaults tty.js@0.2.7
verbose caching /usr/share/adafruit/webide/node_modules/redis/package.json
verbose loadDefaults redis@0.8.2
verbose caching /usr/share/adafruit/webide/node_modules/express/package.json
verbose loadDefaults express@3.1.0
verbose caching /usr/share/adafruit/webide/node_modules/passport-bitbucket/package.json
verbose loadDefaults passport-bitbucket@0.1.2
verbose caching /usr/share/adafruit/webide/node_modules/socket.io/package.json
verbose loadDefaults socket.io@0.9.13
verbose caching /usr/share/adafruit/webide/node_modules/jsDAV/package.json
[... 18 lines cut ...]
verbose url resolved https://registry.npmjs.org/cookie/0.2.3
http GET https://registry.npmjs.org/cookie/0.2.3
ERR! Error: failed to fetch from registry: cookie/0.2.3
ERR!     at /usr/share/npm/lib/utils/npm-registry-client/get.js:139:12
ERR!     at cb (/usr/share/npm/lib/utils/npm-registry-client/request.js:31:9)
ERR!     at Request._callback (/usr/share/npm/lib/utils/npm-registry-client/request.js:136:18)
ERR!     at Request.callback (/usr/lib/nodejs/request/main.js:119:22)
ERR!     at Request.<anonymous> (/usr/lib/nodejs/request/main.js:212:58)
ERR!     at Request.emit (events.js:88:20)
ERR!     at ClientRequest.<anonymous> (/usr/lib/nodejs/request/main.js:412:12)
ERR!     at ClientRequest.g (events.js:156:14)
ERR!     at ClientRequest.emit (events.js:67:17)
ERR!     at HTTPParser.parserOnIncomingClient [as onIncoming] (http.js:1256:7)
ERR! You may report this log at:
ERR!     <http://bugs.debian.org/npm>
ERR! or use
ERR!     reportbug --attach /usr/share/adafruit/webide/npm-debug.log npm
ERR! 
ERR! System Linux 4.1.13+
ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "cookie"
ERR! cwd /usr/share/adafruit/webide
ERR! node -v v0.6.19
ERR! npm -v 1.1.4
ERR! message failed to fetch from registry: cookie/0.2.3
verbose exit [ 1, true ]

stickywes, 2016-02-09:
When npm hits "message failed to fetch" you are most often running into a versioning problem.

Indeed, the 0.6.xxx version of node isn't supported anymore and is quite old.  Wheezy won't update to a supported version of node and npm without a little bit of extra work, too.  I've tried several recommendations at http://stackoverflow.com/a/21715730/1387973 but have not had success.

I recommend avoiding the suggestion that includes `curl -sL https://deb.nodesource.com/setup` because it might interfere with other fixes until you remove the repository that script adds.

The quickest fix for me is to just use a Jessie Raspian distribution instead of trying to unravel the Wheezy issues I'm having.  If being on Wheezy isn't a requirement, then that would be recommendation.

MonadicNomad, 2016-02-09:
The installer should not ruin the existing install.  Ideally it should notify user of issue instead of prompting to upgrade.

judamarto, 2016-04-05:
Got exactly the same issue as @2stacks and @MonadicNomad. Any solution?

agidi, 2016-04-17:
Bumping this, im stuck at the cookie errors

ladyada, 2016-04-17:
hiya - bumping is not going to make this fixed any faster, locking until justin has spare time to review it!

closed 2018-03-06 by jwcooper