adafruit/Adafruit-WebIDE #298: libv8.so.3.14.5 is not available

bezineb5, 2015-12-07:
Hello,

I tried to install the latest version of WebIDE, but it doesn't start. When I try to run /usr/share/adafruit/webide/bin/node_hf/node, I get the following error:
/usr/share/adafruit/webide/bin/node_hf/node: error while loading shared libraries: libv8.so.3.14.5: cannot open shared object file: No such file or directory

In /usr/lib the installed version is:
lrwxrwxrwx  1 root   root      17 Nov  6  2012 libv8.so -> libv8.so.3.8.9.20
-rw-r--r--  1 root   root 2958648 Nov  6  2012 libv8.so.3.8.9.20

It seems that the required version of libv8 for node is not available on raspbian.

Thanks!
Benjamin

uhsmartpowerlab, 2016-02-23:
The latest version of WebIDE (at least version 0.3.12) is packaged with a version of NodeJS that doesn't work on a Raspberry Pi model B. There is some more information and a workaround at https://forums.adafruit.com/viewtopic.php?f=19&t=85839

mfripp, 2016-02-23:
The latest version of WebIDE (at least version 0.3.12) is packaged with a version of NodeJS that doesn't work on a Raspberry Pi model B. There is some more information and a workaround at https://forums.adafruit.com/viewtopic.php?f=19&t=85839

closed 2018-03-06 by jwcooper