adafruit/Adafruit-WebIDE #299: Unable to clone projects or changes

matt65, 2015-12-08:
On my Raspberry Pi, using a fresh install of Raspbian Jessie and with only WebIDE installed after installing WebIDE and logging into my BitBucket account only the Adafruit-Raspberry-Pi-Python-Code repository is available.

I attempted to clone the my-pi-projects from my BitBucket account, after inserting the SSH URL into the WebIDE clone repository pop-up I am getting the error:
Cloning into 'my-pi-projects'... Permission denied (publickey). fatal: Could not read from remote repository. Please make sure you have the correct access rights and the repository exists.

[image]

If I was to follow the BitBucket steps for cloning via a terminal window an empty repository and committing an empty txt file, the file is pushed and that repository is then available within WebIDE.  If I was to edit the file and click save WebIDE is reporting the file has been saved however it won't upload the changes to BB.

Eventually, after saving changes to a file, I will get the error message Error: Failure pushing code to remote repository
[image]

I have changed the owner on the new repository (was previously root:root) so that it is:
drwxr-xr-x  4 webide webide 4096 Dec  8 10:18 PiCar

This matches the existing repositories permissions.

I am running WebIDE v0.3.12, I have done an apt-get update/install/upgrade, installed npm and installed cookie (seen that referenced in a issue the other day).

Can anyone advise how I can resolve this or possibly log files which might give further information?

Your help is greatly appreciated.

tkiener, 2016-01-06:
same problem here...

CheapB, 2016-02-29:
While I had to reinstall to get rid of this issue - there are some other error messages similar to this you will see with an expanded environment, so make sure you are not out of space

phoniclynx, 2016-03-13:
I've done a reinstall hoping it would fix it.. but I'm still getting those same errors.

CayenneSF, 2016-11-22:
Same problem here....

closed 2018-03-06 by jwcooper