adafruit/Adafruit-WebIDE #306: Install problem - Raspbian jessie - nodejs:  conflicts: nodejs-legacy

julesbike, 2016-04-11:
The install script crashes with the following error:

Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 nodejs : Conflicts: nodejs-legacy
E: Unable to correct problems, you have held broken packages.

Any ideas??

ratnick, 2016-06-16:
Hi, I have the same problem on a Beaglebon Black with a completely fresh installation of Debian 8.4. (Jessie). All I have done so far is to:
1. install Debian 8.4 on an SD card (so not flashing to eMMC)
2. Enlarge the partition on the SD card to 16Gb
3. Set the timezone
4. Installed AdaFruit
      # apt-get install build-essential python-dev python-setuptools python-pip python-smbus   
      # pip install Adafruit_BBIO
5. Tried to install the WEBIde:
      # curl https://raw.githubusercontent.com/adafruit/Adafruit-WebIDE/alpha/scripts/install.sh | sudo sh

But then I get:
The following packages have unmet dependencies:
 nodejs : Conflicts: nodejs-legacy
E: Unable to correct problems, you have held broken packages.

I have absolutely no clue about this, but it's preventing me from coding. Is there a good alternative to Adafruit. I have Python code I'd like to run and debug.
I have spent nearly 12 hours on this now :-(

The-Alchemist, 2016-07-28:
The workaround is to change the line
````
apt-get install nodejs nodejs-legacy ......
````

Remove ``nodejs``, so you only have ``nodejs-legacy`` in the list of packages.

ratnick, 2016-07-28:
Hi again,
Thanks for the input. After struggling for a looong time back in June, I decided to scrap WEBide and use another, windows based IDE (Wing IDE). That works very well - better than WEBide, since it debugs multi-threaded applications like a charm.
But thanks anyway - your suggestion may help others.

koustabhdolui, 2017-05-16:
Hi @The-Alchemist, it should be the other way round. nodejs-legacy depends on nodejs. So it makes sense to just keep nodejs and remove nodejs-legacy. Thank you for the pointer, worked for me :)

motaba1, 2017-12-04:
I have same problem. Raspberry Pi 3 with Raspbian (Linux 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux).
Tried to install the WEBIde: curl https://raw.githubusercontent.com/adafruit/Adafruit-WebIDE/alpha/scripts/install.sh | sudo sh
Same error:
**** Installing required libraries ****
**** (redis-server git restartd libcap2-bin avahi-daemon i2c-tools python-smbus) ****
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 http://mirrordirector.raspbian.org/raspbian stretch InRelease
Hit:3 http://ppa.launchpad.net/aziotsdklinux/ppa-azureiot/ubuntu vivid InRelease
Hit:4 https://deb.nodesource.com/node_4.x stretch InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
avahi-daemon is already the newest version (0.6.32-2).
git is already the newest version (1:2.11.0-3+deb9u2).
i2c-tools is already the newest version (3.1.2-3).
libcap2-bin is already the newest version (1:2.25-1).
python-smbus is already the newest version (3.1.2-3).
nodejs is already the newest version (4.8.6-1nodesource1).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

**The following packages have unmet dependencies:
 nodejs : Conflicts: nodejs-legacy
E: Unable to correct problems, you have held broken packages.**

I don't have installed nodejs-legacy.
sudo apt-get remove --purge nodejs-legacy
Package 'nodejs-legacy' is not installed, so not removed

My nodejs is:
nodejs_4.8.6-1nodesource1_armhf.deb
sudo apt-get install nodejs
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  nodejs
0 upgraded, 1 newly installed, 0 to remove and 1 not upgraded.
Need to get 0 B/7,468 kB of archives.
After this operation, 33.7 MB of additional disk space will be used.
Selecting previously unselected package nodejs.
(Reading database ... 135218 files and directories currently installed.)
Preparing to unpack .../nodejs_4.8.6-1nodesource1_armhf.deb ...
Unpacking nodejs (4.8.6-1nodesource1) ...
Setting up nodejs (4.8.6-1nodesource1) ...
Processing triggers for man-db (2.7.6.1-2) ...

Can you help me?

andremsousa, 2017-12-12:
Getting the same error.
I remember installing this on Raspberry Pi 1, now on 3 it does not work.
@The-Alchemist how can i remove that line?
Where is the folder?
Thanks!

closed 2018-03-06 by jwcooper