adafruit/Adafruit-WebIDE #319: Can't connect from browser

jtiscione, 2017-06-04:
I ran webide.sh to install on a Raspberry pi 3, and it generated the following output:

**** Starting the server...(please wait) ****
**** The Adafruit WebIDE is installed and running! ****
**** Commands: sudo service adafruit-webide.sh {start,stop,restart} ****
**** Navigate to http://pi3.local to use the WebIDE

The Raspberry Pi refuses all browser connections, at least on port 80.

How do I uninstall this thing?

CheapB, 2018-01-30:
Are you on Stretch  or Jessie? I cant get this to work on a clean Stretch install but got the same issues you describe. works well under Jessie

stemalda, 2018-02-23:
I have the same problem running Stretch ....

arcasoy, 2018-03-05:
Did anyone find a fix to this? Whenever I try to connect it does the same thing, not picking up the connection at all. What to do? I am also using UbuntuMATE

jwcooper, 2018-03-06:
The webide is being worked on in a 'rework' branch (pull request above). We're simplifying the webide, and hoping it will be easier to support in the future. You could try the new branch as we work on it.

Thanks!

closed 2018-03-06 by jwcooper