adafruit/Adafruit-WebIDE #325: npm install gives errors

slarionoff, 2017-12-02:
Dear colleagues!

I just finished clean install of Raspbian (2017-09-07-raspbian-stretch), then made a dist-upgrade and then started a Manual webIDE installation (like it is said here: https://learn.adafruit.com/webide/installation). All steps before `npm install` went smoothly, but this step gave me some errors. I'm not a specialist in Node.js, could you please explain what is missing at my system? Googling didn't help.

`pi@raspberrypi:~/Python/Adafruit-WebIDE $ npm install`
`npm WARN package.json gitty@0.0.1 license should be a valid SPDX license expression`
`npm WARN package.json later@0.0.5 No license field.`
`npm WARN package.json passport-github@0.1.3 No README data`
`npm WARN package.json passport-github@0.1.3 No license field.`
`npm WARN package.json jsDAV@0.1.4 No license field.`
`npm WARN package.json connect@2.7.3 No license field.`
`npm WARN package.json express@3.1.0 No license field.`
`npm WARN package.json jade@0.28.1 No license field.`
`npm WARN package.json passport@0.1.15 No license field.`
`npm WARN package.json passport-bitbucket@0.1.2 No license field.`
`npm WARN package.json redis@0.8.2 No license field.`
`npm WARN package.json request@2.14.0 No license field.`
`npm WARN package.json socket.io@0.9.13 No license field.`
`npm WARN package.json validator@0.4.25 No license field.`
`npm WARN package.json winston@0.6.2 No license field.`
`npm WARN deprecated connect@2.7.2: connect 2.x series is deprecated`
`-`
`> pty.js@0.2.13 install /home/pi/Python/Adafruit-WebIDE/node_modules/tty.js/node_modules/pty.js`
`> node-gyp rebuild`
``
`make: Entering directory '/home/pi/Python/Adafruit-WebIDE/node_modules/tty.js/node_modules/pty.js/build'`
`  CXX(target) Release/obj.target/pty/src/unix/pty.o`
`In file included from ../src/unix/pty.cc:20:0:`
`../node_modules/nan/nan.h:324:27: error: redefinition of ‘template<class T> v8::Local<T> Nan::imp::NanEnsureHandleOrPersistent(const v8::Local<T>&)’`
`   NAN_INLINE v8::Local<T> NanEnsureHandleOrPersistent(const v8::Local<T> &val) {`
`                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~`
`../node_modules/nan/nan.h:319:17: note: ‘template<class T> v8::Handle<T> Nan::imp::NanEnsureHandleOrPersistent(v8::Handle<T>&)’ previously declared here`
`   v8::Handle<T> NanEnsureHandleOrPersistent(const v8::Handle<T> &val) {`
`                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~`
`../node_modules/nan/nan.h:344:27: error: redefinition of ‘template<class T> v8::Local<T> Nan::imp::NanEnsureLocal(v8::Handle<T>&)’`
`   NAN_INLINE v8::Local<T> NanEnsureLocal(const v8::Handle<T> &val) {`
`                           ^~~~~~~~~~~~~~`
`../node_modules/nan/nan.h:334:27: note: ‘template<class T> v8::Local<T> Nan::imp::NanEnsureLocal(const v8::Local<T>&)’ previously declared here`
`   NAN_INLINE v8::Local<T> NanEnsureLocal(const v8::Local<T> &val) {`
`                           ^~~~~~~~~~~~~~`
`../node_modules/nan/nan.h:757:13: error: ‘node::smalloc’ has not been declared`
`     , node::smalloc::FreeCallback callback`
`             ^~~~~~~`
`../node_modules/nan/nan.h:757:35: error: expected ‘,’ or ‘...’ before ‘callback’`
`     , node::smalloc::FreeCallback callback`
`                                   ^~~~~~~~`
`../node_modules/nan/nan.h: In function ‘v8::Local<v8::Object> NanNewBufferHandle(char*, size_t, int)’:`
`../node_modules/nan/nan.h:761:50: error: ‘callback’ was not declared in this scope`
`         v8::Isolate::GetCurrent(), data, length, callback, hint);`
`                                                  ^~~~~~~~`
`../node_modules/nan/nan.h:761:60: error: ‘hint’ was not declared in this scope`
`         v8::Isolate::GetCurrent(), data, length, callback, hint);`
`                                                            ^~~~`
`../node_modules/nan/nan.h: In function ‘v8::Local<v8::Object> NanNewBufferHandle(const char*, uint32_t)’:`
`../node_modules/nan/nan.h:768:67: error: no matching function for call to ‘New(v8::Isolate*, const char*&, uint32_t&)’`
`     return node::Buffer::New(v8::Isolate::GetCurrent(), data, size);`
`                                                                   ^`
`In file included from ../node_modules/nan/nan.h:25:0,`
`                 from ../src/unix/pty.cc:20:`
`/usr/include/nodejs/src/node_buffer.h:34:40: note: candidate: v8::MaybeLocal<v8::Object> node::Buffer::New(v8::Isolate*, v8::Local<v8::String>, node::encoding) <near match>`
` NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,`
`                                        ^~~`
`/usr/include/nodejs/src/node_buffer.h:34:40: note:   conversion of argument 3 would be ill-formed:`
`In file included from ../src/unix/pty.cc:20:0:`
`../node_modules/nan/nan.h:768:67: error: invalid conversion from ‘uint32_t {aka unsigned int}’ to ‘node::encoding’ [-fpermissive]`
`     return node::Buffer::New(v8::Isolate::GetCurrent(), data, size);`
`                                                                   ^`
`In file included from ../node_modules/nan/nan.h:25:0,`
`                 from ../src/unix/pty.cc:20:`
`/usr/include/nodejs/src/node_buffer.h:46:40: note: candidate: v8::MaybeLocal<v8::Object> node::Buffer::New(v8::Isolate*, char*, size_t) <near match>`
[... 78 lines cut ...]
`17 warn package.json redis@0.8.2 No license field.`
`18 warn package.json request@2.14.0 No license field.`
`19 warn package.json socket.io@0.9.13 No license field.`
`20 warn package.json validator@0.4.25 No license field.`
`21 warn package.json winston@0.6.2 No license field.`
`1303 warn deprecated connect@2.7.2: connect 2.x series is deprecated`
`5111 warn This failure might be due to the use of legacy binary "node"`
`5112 warn For further explanations, please read`
`/usr/share/doc/nodejs/README.Debian`
`5159 error pty.js@0.2.13 install: `node-gyp rebuild``
`5159 error Exit status 1`
`5160 error Failed at the pty.js@0.2.13 install script.`
`5160 error This is most likely a problem with the pty.js package,`
`5160 error not with npm itself.`
`5160 error Tell the author that this fails on your system:`
`5160 error     node-gyp rebuild`
`5160 error You can get their info via:`
`5160 error     npm owner ls pty.js`
`5160 error There is likely additional logging output above.`
`5161 error System Linux 4.9.59+`
`5162 error command "/usr/bin/nodejs" "/usr/bin/npm" "install"`
`5163 error cwd /home/pi/Python/Adafruit-WebIDE`
`5164 error node -v v4.8.2`
`5165 error npm -v 1.4.21`
`5166 error code ELIFECYCLE`

slarionoff, 2017-12-03:
And yeah, I've checked the path from the very beginning (install /home/pi/Python/Adafruit-WebIDE/node_modules/tty.js/node_modules/pty.js), there's no such folder, only /home/pi/Python/Adafruit-WebIDE/node_modules/ plus some sub-folders. There's no tty.js file/folder there.

h4ck0rm4n, 2017-12-04:
Same Problem here...
I havnt found a solution by now

slarionoff, 2017-12-20:
Well, open-source is such a thing - if you want something, do it yourself :-). So it seems that I solved this issue. Below are the steps I passed:
1. I created a copy of `https://github.com/jwcooper/tty.js` repository at my account, `https://github.com/slarionoff/tty.js`;
2. Changed there a file package.json, line 17 from `"pty.js": "0.2.13"` to `"pty.js": ">=0.2.13"`;
3. Changed a local RPi file package.json, line 21 from `"tty.js": "git://github.com/jwcooper/tty.js.git",` to  `"tty.js": "git://github.com/slarionoff/tty.js.git",`
4. Started `npm install`;
5. Installation finished successfully and I was able to connect to a Bit-bucket;

So far so good. In case of issues I'll report them here.

closed 2017-12-20 by slarionoff