adafruit/Adafruit-WebIDE #326: Installation issue, with temporary fix(?)

JoeSpinogatti, 2017-12-27:
Sorry if this isn't the right place to put this, but just wanted to add it to the official list in case it's helpful to anyone else after a couple hours of searching. 

I tried to install and uninstall the WebIDE multiple times and kept getting an error in the browser that I couldn't connect (conenction refused) no matter what port it was on. Tried both manual and script installations - no dice. 

Finally reinstalled from the script and followed the suggestion from the following forum to get it to work. Scroll down to post by mfripp with commands.

https://forums.adafruit.com/viewtopic.php?f=19&t=85839

JoeSpinogatti, 2017-12-27:
Update: After trying to work with this for a little bit, it gets things to run/install, but it is essentially unusable in this state.

CheapB, 2018-01-30:
Are you on Stretch or Jessie?

jwcooper, 2018-03-05:
We're working on simplifying the webide. There is a beta branch you can try out, if you want to tinker with it. Most likely you'll run into bugs, but it should be in a more usable state than the master branch is now.

closed 2018-03-05 by jwcooper