adafruit/Adafruit-WebIDE #328: npm ERR! Error: EPERM: operation not permitted, chown '/home/webide/.npmrc'

Darioits, 2018-04-25:
pi@raspberrypi:~ $ su

Password:
root@raspberrypi:/home/pi# curl https://raw.githubusercontent.com/adafru...install.sh | sudo sh
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  3758  100  3758    0     0   9348      0 --:--:-- --:--:-- --:--:--  9371
false
**** Downloading the latest version of the WebIDE ****
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   131    0   131    0     0    190      0 --:--:-- --:--:-- --:--:--   190
100 2379k    0 2379k    0     0   414k      0 --:--:--  0:00:05 --:--:--  516k
**** Installing required libraries ****
**** (nodejs-legacy npm git i2c-tools python-smbus ntp libkrb5-dev) ****
Hit:1 http://repo.mosquitto.org/debian stretch InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Get:3 http://archive.raspberrypi.org/debian stretch InRelease [25.3 kB]
Get:4 http://archive.raspberrypi.org/debian stretch/ui armhf Packages [30.7 kB]
Fetched 71.0 kB in 1s (44.1 kB/s)
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
git is already the newest version (1:2.11.0-3+deb9u2).
i2c-tools is already the newest version (3.1.2-3).
libkrb5-dev is already the newest version (1.15-1+deb9u1).
nodejs-legacy is already the newest version (4.8.2~dfsg-1).
npm is already the newest version (1.4.21+ds-2).
ntp is already the newest version (1:4.2.8p10+dfsg-3+deb9u2).
python-smbus is already the newest version (3.1.2-3).
The following packages were automatically installed and are no longer required:
  dnsutils echoping libconfig-grammar-perl libjs-cropper libjs-prototype libjs-scriptaculous libsnmp-session-perl
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
**** Create webide user and group ****
groupadd: group 'webide' already exists
useradd: user 'webide' already exists
**** Adding webide user to sudoers ****
/etc/sudoers.tmp: parsed OK
/etc/sudoers.d/010_pi-nopasswd: parsed OK
/etc/sudoers.d/README: parsed OK
**** Installing webide dependencies ****
npm ERR! Error: EPERM: operation not permitted, chown '/home/webide/.npmrc'
npm ERR!     at Error (native)
npm ERR!  { [Error: EPERM: operation not permitted, chown '/home/webide/.npmrc']
npm ERR!   errno: -1,
npm ERR!   code: 'EPERM',
npm ERR!   syscall: 'chown',
npm ERR!   path: '/home/webide/.npmrc' }
npm ERR!
npm ERR! Please try running this command again as root/Administrator.

npm ERR! System Linux 4.14.34-v7+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "config" "set" "tmp" "/home/webide/tmp"
npm ERR! cwd /usr/share/adafruit/webide
npm ERR! node -v v4.8.2
npm ERR! npm -v 1.4.21
npm ERR! path /home/webide/.npmrc
npm ERR! syscall chown
npm ERR! code EPERM
npm ERR! errno -1
npm ERR! stack Error: EPERM: operation not permitted, chown '/home/webide/.npmrc'
npm ERR! stack     at Error (native)
npm ERR!
npm ERR! Additional logging details can be found in:
npm ERR!     /usr/share/adafruit/webide/npm-debug.log
npm ERR! not ok code 0

jwcooper, 2018-04-25:
Try running this on your home:

```chown -R webide:webide /home/webide```

Appears that not everything in there is owned by the webide user and group.

Was there an already existing installation on your Pi that you upgraded? Who currently owns the .npmrc dir?

Darioits, 2018-04-26:
output generate a very long list of warning

pi@raspberrypi:~ $ chown -R webide:webide /home/webide
chown: changing ownership of '/home/webide/.npm/basic-auth/2.0.0/package.tgz': Operation not permitted
chown: changing ownership of '/home/webide/.npm/basic-auth/2.0.0/package/package.json': Operation not permitted
chown: changing ownership of '/home/webide/.npm/basic-auth/2.0.0/package': Operation not permitted
chown: changing ownership of '/home/webide/.npm/basic-auth/2.0.0': Operation not permitted
chown: changing ownership of '/home/webide/.npm/basic-auth': Operation not permitted

pi@raspberrypi:~ $ ls -alt /home/webide/
total 36
-rw-r--r--   1 webide webide    23 Apr 24 09:36 .npmrc
drwxr-xr-x 147 webide webide 20480 Apr 24 09:34 .npm
drwxr-xr-x   3 webide webide  4096 Apr 24 09:33 tmp
drwxr-xr-x   4 webide webide  4096 Apr 24 09:33 .
drwxr-xr-x   4 root   root    4096 Apr 21 08:42 ..

jwcooper, 2018-04-26:
Oh, sorry, you'd need to run it like this:

```sudo chown -R webide:webide /home/webide```

I'm not able to reproduce your issue yet. Still looking into it.

Darioits, 2018-04-27:
ok  thank you, I run your command and restart install processes
same problem

pi@raspberrypi:~ $ cat /usr/share/adafruit/webide/npm-debug.log
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/nodejs',
1 verbose cli   '/usr/bin/npm',
1 verbose cli   'config',
1 verbose cli   'set',
1 verbose cli   'tmp',
1 verbose cli   '/home/webide/tmp' ]
2 info using npm@1.4.21
3 info using node@v4.8.2
4 info config set "tmp" "/home/webide/tmp"
5 error Error: EPERM: operation not permitted, chown '/home/webide/.npmrc'
5 error     at Error (native)
5 error  { [Error: EPERM: operation not permitted, chown '/home/webide/.npmrc']
5 error   errno: -1,
5 error   code: 'EPERM',
5 error   syscall: 'chown',
5 error   path: '/home/webide/.npmrc' }
6 error Please try running this command again as root/Administrator.
7 error System Linux 4.14.34-v7+
8 error command "/usr/bin/nodejs" "/usr/bin/npm" "config" "set" "tmp" "/home/webide/tmp"
9 error cwd /usr/share/adafruit/webide
10 error node -v v4.8.2
11 error npm -v 1.4.21
12 error path /home/webide/.npmrc
13 error syscall chown
14 error code EPERM
15 error errno -1
16 error stack Error: EPERM: operation not permitted, chown '/home/webide/.npmrc'
16 error stack     at Error (native)
17 verbose exit [ -1, true ]

In history I see also this error:
 964  chown: changing ownership of '/home/webide/.npm/basic-auth/2.0.0/package.tgz': Operation not permitted
  965  chown: changing ownership of '/home/webide/.npm/basic-auth/2.0.0/package/package.json': Operation not permitted
  966  chown: changing ownership of '/home/webide/.npm/basic-auth/2.0.0/package': Operation not permitted
  967  chown: changing ownership of '/home/webide/.npm/basic-auth/2.0.0': Operation not permitted
  968  chown: changing ownership of '/home/webide/.npm/basic-auth': Operation not permitted

Snowcandle, 2018-04-29:
I'm getting the exact same text trying to install. I reinstalled raspbian and it keeps happening.

regearhead, 2018-04-29:
I'm also getting this error message on install.

regearhead, 2018-04-29:
Also found another user with this issue: 
https://forum.raspberryitaly.com/showthread.php?tid=3060

jwcooper, 2018-05-01:
I just pushed a new version of the install script. I tested on a fresh build of debian stretch, and all seems to work OK now.

Thanks for the bug report!

closed 2018-05-01 by jwcooper