adafruit/Adafruit-WebIDE #339: Last couple of lines cut off "editor"

kkalbaugh, 2018-09-22:
The editor won't scroll down to the last few lines of the code.   Normally cuts off about the last 3 lines.

- Windows 7, tried Chrome (ver 69), Firefox (ver 62) and IE 11
- Normal  (100%) Windows Font Size
- Same result on both 1920x1080 and 1680x1050 monitors
- No zoom set in browsers
- I tried different font sizes in the IDE settings

Here's a screenshot:
[image]

This project is great!  It's very helpful when you're playing around and making frequent code changes.

dylanpoll, 2020-05-07:
this is caused by zooming into the browser. if you run it at 100% view it shouldn't be an issue.

dylanpoll, 2020-05-07:
if you zoom out you can still see the information, it is not fun though. but the dev team for this I'm sure has a lot of work and its a free program.