adafruit/Adafruit-WebIDE #343: Clone Repository Not Working

jxsdev, 2019-05-01:
With a fresh install on a Raspberry Pi 2B I can't get any Git functionality to work.

Clicking + Clone Repository always results in "TypeError: Path must be a string. Received undefined"

After running a debugger, it looks like this 'req.user' being passed as the repository_url to clone_update_remote_push.

master/controllers/editor.js:
145 git_helper.clone_update_remote_push(req.user, repository_url, function(err, status) {

master/helpers/git_helper.js:
82 exports.clone_update_remote_push = function(repository_url, cb) {

ronaldroe, 2019-08-18:
Same issue here.

13967585277, 2019-09-09:
So am I。Wish to solve it quickly

AndreasPercher, 2020-05-10:
Same with  git@gitlab.com:<ourcompany>/<my project>.git

Any Help?