adafruit/Adafruit-WebIDE #347: Installer outdated and fails on Raspbian Buster

jason0031, 2020-07-10:
On a Raspberry Pi 2B with Raspbian Buster I tried to installer:
 
`curl` https://raw.githubusercontent.com/adafruit/Adafruit-WebIDE/master/scripts/install.sh | sudo sh`

After getting a warning "Node.js 8.x LTS Carbon is no longer actively supported!" the installer continues but will fail with:

**sh: 68: npm: not found**

The installer script is outdated and fails.

ThibaultNocchi, 2020-07-10:
Node isn't being installed as the script is trying to installed an outdated version. Maybe replace the line:

```bash
curl -sL https://deb.nodesource.com/setup_8.x | sudo -E bash -
```

by

```bash
curl -sL https://deb.nodesource.com/setup_12.x | sudo -E bash -
```

SpacerDave, 2021-01-20:
Thanks for suggestion above.  I've tried to install WebIDE using the standard link to 'install.sh' as described in the documentation.  I'm running 'buster' and also found the 'sh: 68: npm: not found' issue.  I copied the install.sh script and changed the 'setup_8.x' to 'setup_12.x' as suggested.  Saved it and chmod +x and ran via command 'sudo bash install.sh'.  During Node.js install, I get the error listed below and it crashes.  Any further suggestions?  Should I install/revert to 'Jessie' or 'Stretch'? This used to run on my Pi 3B, but this was back in 2017 and I was likely running Jessie at the time.  

## Confirming "buster" is supported...
+ curl -sLf -o /dev/null 'https://deb.nodesource.com/node_12.x/dists/buster/Release'
## Your distribution, identified as "buster", is not currently supported, please contact NodeSource at https://github.com/nodesource/distributions/issues if you think this is incorrect or would like your distribution to be considered for support